DalFSAE/battery-management-system #12: Define BMS software architecture and functional requirements

AshtonDudley, 2025-08-22:


AshtonDudley, 2025-09-05:
Also see notes added here: https://github.com/DalFSAE/battery-management-system/issues/11#issuecomment-3256961036

AshtonDudley, 2025-10-03:
> Can we get write access to repos to push branch and create PR: $ git push origin Task12_BMS_Software_Architecture_and_Requirements remote: Permission to DalFSAE/battery-management-system.git denied to EytanME.

For future reference, I figured out the issue. The changes were made on a fork of the repo, which has separate permissions for merging. I added the files to #25, so we can just work off that branch going forward

nheinstein, 2025-10-20:
I have been working on the Functional Requirements for the BMS. I am like 75%/80% done them so far. I am planning on meeting up with @ahmed18007 sometime this week.

nheinstein, 2025-10-28:
@Lorne-Mac, I have made a pull request with my MD documentation for the functional requirements. Me and @ahmed18007 are going to meet sometime this week, to work on finishing up the software architecture.